CCHS-Melbourne/iotuz-esp32-hardware #5: Bigger drills

mage0r, 2016-11-29:
For speaker (just fits).
For rotary encoder (doesn't fit signal pins)
APA106 (increase spacing)

mage0r, 2016-11-29:
Fixed spacing on APA106 footprint in 2107f38

mage0r, 2016-11-29:
Fixed hole sizes for encoder in 8cdbe23

rdpowers, 2016-11-30:
Add bigger pads around TH parts where possible to make hand-soldering easier. It's pretty small on the speaker, encoder, IRTx, & APA106. Wouldn't hurt to do it on all of the TH stuff though.

mage0r, 2016-12-01:
I've increased the pads on the speaker and the rotary encoder ones are already fairly big.  I can't do much about the apa106, it's pins are fairly close together and if I make the pads much larger we'll have problems with bridging.

rdpowers, 2016-12-03:
Did you increase the spacing of the APA106 pins? On second thought, this is probably not a good idea - they probably should have been closer if possible, or left alone if not.

mage0r, 2016-12-03:
I reduced them, but I haven't updated the PCB.  Same with the other footprints, They've been corrected in the kicad_mod files, but the change hasn't gone through the PCB yet.

mage0r, 2016-12-05:
Updated the pcb in e62c2e9

closed 2016-12-05 by mage0r