CCHS-Melbourne/iotuz-esp32-hardware #30: X & Y Pinout on Joystick Reversed

rdpowers, 2017-01-05:
Minor oversight in pinout of joystick. X pins are Y pins and Y pins are X pins.

mage0r, 2017-01-05:
Are you sure this is still an issue?

At one point I switched the x and y over because the joystick was being mounted 90 degrees off on the board.

rdpowers, 2017-01-05:
Yes. In reality, the X pot is aligned along the Y axis and vice-versa. Takes a second to realise that actuation means it needs to rotate about the tangent axis so intuition about what's X and what's Y is not normal.
  
[image]

Also needs updating in documentation, so probably leave this open until that's fixed up as well.

mage0r, 2017-01-05:
Hrm, I definitely re-labelled something to try and fix that.

It might have been the labels on the schematic.  but then, they're just text labels so they could have been fixed back up.

Is the pin list on the back wrong?

rdpowers, 2017-01-05:
Yes, that is incorrect also in the same way.