CCHS-Melbourne/iotuz-esp32-hardware #44: Adafruit_ILI9341.cpp doesn't work without patch

marcmerlin, 2017-01-17:
I've compiled graphicstest.ino and it fails to build 
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp: In member function 'void Adafruit_ILI9341::spiwrite(uint8_t)':
/home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp:113:3: error: 'mosiport' was not declared in this scope
  *mosiport |=  mosipinmask;
   ^

In Adafruit_ILI9341.cpp:96, I added a single define to select the correct code, and then everything works:
#define ESP8266
#if defined(ESP8266) || defined (ARDUINO_ARCH_ARC32)

I used the arduino installer for the lib and installed lib 1.0.1 and it didn't work until I patched it.
Not sure if the board file for ESP32 is supposed to define  ESP8266 or if the adafruit lib is supposed to detect a #define ESP32

marcmerlin, 2017-01-17:
I also updated my arduino IDE from 1.6.7 to 1.8.1 and it didn't help, same problem, but the workaround still works.

projectgus, 2017-01-17:
The ESP32 board file can't safely pretend to be an ESP8266, because it isn't one! Lots of system-level APIs are different so some Arduino libraries would error out / crash.

I think eventually the popular Arduino libraries will be all updated to support ESP32. The hard thing about some of these libraries is they're not hardware agnostic, so someone has to add the relevant support

I expect if you send Adafruit a pull request that said "I needed this `|| defined(ESP32)` for ESP32 Arduino support and it worked afterwards", then they'd accept it.

marcmerlin, 2017-01-17:
That sounds more than reasonable, except it works for others without this patch, but I'm worried I have something wrong on my side. Once we figure out that the adafruit library is indeed in need of that patch, I'll send it, I just would like to know why it works for others without the patch :)

mage0r, 2017-01-17:
I can confirm the same problem.  I fixed mine using the method suggested by Angus.

marcmerlin, 2017-01-17:
Looks like it's already upstream, just not in the released version?
https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L98

marcmerlin, 2017-01-17:
Added warning in the code
https://github.com/CCHS-Melbourne/iotuz-esp32-hardware/pull/45

closed 2017-01-17 by marcmerlin