CCHS-Melbourne/iotuz-esp32-hardware #46: Are APA106 LEDs missing recommended cap? Has anyone gotten them to work?

marcmerlin, 2017-01-17:
Looking at the top right of the first page of https://github.com/CCHS-Melbourne/iotuz-esp32-hardware/raw/master/Circuit/Resources/IoTuz-Schematic.pdf (took me a while to locate them :) ), they do not seem to have any cap.

https://learn.adafruit.com/adafruit-neopixel-uberguide/individual-neopixels says 
"It’s very strongly recommended that each NeoPixel have an accompanying 0.1 μF capacitor between +V and ground. This prevents communication problems due to brownout of the on-pixel driver logic. It’s occasionally sufficient to have one capacitor between pairs of pixels; some of our NeoPixel rings work that way."

Is that an issue for the design in this board?
More generally has anyone managed to talk to them with the neopixel library?
So far I have
#include "Adafruit_NeoPixel.h"
#define RGB_LED_PIN 23
#define NUMPIXELS 2
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, RGB_LED_PIN, NEO_GRB + NEO_KHZ800);
pixels.begin();
pixels.setPixelColor(0, pixels.Color(255, 0, 0));
pixels.setPixelColor(1, pixels.Color(0, 255, 0));
pixels.show();

And nothing happens.
I've also noticed that sometimes (rarely) when you boot/power the board, the LEDs come with a different color, meaning that noise comes to them and get taken as a command.

wolfeidau, 2017-01-17:
I have been looking into the software side of this https://github.com/FastLED/FastLED supports the apa106 but doesn't have ESP32 support. We are discussing forking it and working with @projectgus to get it working.

Will keep you updated!

rdpowers, 2017-01-21:
As probably seen, this has been solved in software. 

@marcmerlin is right that adhering to the datasheet is a good idea even when you know you can get away with it otherwise. Tagging as enhancement for next revision.

marcmerlin, 2017-06-06:
just found this old bug. The neopixel library wasn't working because it didn't have any code at all for ESP32. I added some, so now it does :)
Whether the cap is actually necessary for a couple of pixels, now that I'm thinking about it again, the answer is likely no, so feel free to close if you wish.