CCHS-Melbourne/iotuz-esp32-hardware #53: io expander is listed as having 2 IO ports on the back of the board

marcmerlin, 2017-01-23:
@rdpowers, the back of the board mentions: 0x20 and 0x40
The spec sheet seems to imply (from what I read) that if 0x20 is the write port, 0x21 is the read port, and yet the version we have seem to work with a single I/O port, or at least does not have 2 IO ports that I was able to find from any scan or anything else.

Assuming that indeed that expander only has a single IO port, the other one should be removed from the back of the board

rdpowers, 2017-01-24:
I think there's some confusion due to the strangeness encountered in 7-bit versus 8-bit addressing on i2c busses. There is only one device address - 0x20. This is the 7-bit address 0x20 = 010 | 0000 (note the '|' is showing where the separator is and that the upper nibble only has 3 bits). This is how you will access using the Wire library in Arduino. 

Basically that library hides the convention that the last of 8 bits is used for a read/write flag. When you do Wire.read(), the library is taking the 7-bit address and appending a '0' as least-significant-bit. So you end up with a read address of, wait for this, 0x40 = 0100 | 0000. What happens, quite stupidly, is that the convention for referring to i2c address says you move the nibble-separator to where you would expect it to be for 8 bits (now the top nibble has 4 bits!! insane!) and introduce a pile of confusion. Similarly for the Wire.write() address of 0x41 = 0100 | 0001.

Datasheet for the PCF8574 on page 5 (7.1.1 Address maps) might make a little more sense than what I just wrote.. we've tied the address lines to all low, so the first line of Table 4 corresponds to our situation.

I don't think there's a reference to 0x40 on the v1.0 board (am I missing it? I am very bad at finding things). Maybe you're seeing "IO Exp irpt: IO4"? It is difficult to read and could be confused for 40, especially as that is coincident with the 8-bit "write" i2c address (0100 | 0000) for the IO expander.

Clear as mud?

marcmerlin, 2017-01-25:
@rdpowers  thanks for the detailled answer, I had no idea that the wire library would magically change the IO address and that this is why things worked.
I did see 0x40 written somewhere, but I can't find it anymore, so it must have been online and must have been fixed already.
Either way, now I understand how that IO expander is actually supposed to work, and I'll write more comments in my example code so that it's clear for others too.
Thanks much

closed 2017-01-25 by marcmerlin