CCHS-Melbourne/iotuz-esp32-hardware #67: Pins 37/38 are unused, high res input only, use one of them for battery and free up GPIO2

marcmerlin, 2017-03-23:
GPIO 2 could be better used for other things like https://github.com/CCHS-Melbourne/iotuz-esp32-hardware/issues/66

rdpowers, 2017-04-20:
GPIO 37/38 (SENSOR_CAPP/SENSOR_CAPN) are not available on the WROOM module, so this is not feasible.

closed 2017-04-20 by rdpowers

marcmerlin, 2017-04-20:
Ah, I had no idea. Thanks for that info. So basically there are 0 free ports on that chip. Someone not using the I2S support can probably re-use some of the ports, but that's it.
As extra credit, it would interesting to have a list of ports by port number, and show which ones could be re-used with either input or output if you ignore/not use their original purpose.
For instance, let's say port 2 would be re-used as output (maybe?) if you forgo its original use of a battery monitor (since that function isn't working anyway due to how unreliable analogread seems to be at this time).