CCHS-Melbourne/iotuz-esp32-hardware #68: flashing over gives CRC errors that will not go away unless entire board is power cycled, or LCD power is removed

marcmerlin, 2017-03-23:
This depends on what you flash, and maybe whether the LCD is on and what's it's displaying at the time you're flashing (not positive on that).
Seems like an inductance problem.

example error:
Wrote 433536 bytes (199673 compressed) at 0x00010000 in 3.1 seconds (effective 1128.5 kbit/s)...
File  md5: 0792ec26eccb61687ea28d9813277886
Flash md5: 5c6aa8fb63042f23e65f3dd6b174fa72
MD5 of 0xFF is c6525a1270c88fcc86112d95347adaf8

rdpowers, 2017-04-20:
I haven't been able to replicate this on 3 different boards. Needs more investigation. Is there a specific binary that is more/less problematic? I'm intrigued.

marcmerlin, 2017-04-20:
I think it was dependent on what I was running at the time on the board before flashing to it.
I ended up "fixing" this by bending the LCD power pin and connecting it to a wire and a switch.
Strangely, just doing so, seemed to have change the inductance of the circuit in a way that the flashing problem went away and I never had to flip the switch to off.
This was not a one off with my board because the previous board I had, had the same issue, and the board that was loaned to me during LCA, also had he same issue.
I however have a replacement LCD I had to buy off ebay, it's pin compatible but not the same than the broken one, and that one does not seem to have the problem (or the problem went away on its own, or for a different reason).
I think you got my IoTuz code to run, you could run that, selecte demosauce to load up the CPU and LCD, and then try to fly something else from arduino.
If that works, then I'm not sure what to say.