CCHS-Melbourne/iotuz-esp32-hardware #69: BME280 placement causes temp readings 10C too high due to chip placement?

marcmerlin, 2017-03-23:
When I booted my board, it showed 19C which was just about right.
15mn later once the board has run for a while, things warm up, and now it's showing 28C
Yesterday, I put the board (running) in the freezer and got a temp of 5C.
It seems that the board runs warm enough to affect the temp sensor.

marcmerlin, 2017-03-23:
Ok, I did the freezer experiment again but this time I turned the board off. Then I took it out and turned it on right away.  I got a temperature of -0.89C  It feels too warm for a freezer, but confirm again that the board running is definitely raising the board temperature in a measureable way.

rdpowers, 2017-04-20:
Ok, interesting. The normal usage would have cutouts and plane/fill removal around the BME280 to prevent this. I suspected that, given the area, the thermal mass of the PCB would be sufficient to counteract such heating. 5-10C is pretty significant, so this should be mitigated if possible in the next revision.

marcmerlin, 2017-04-20:
run my iotuz code on your board and give it a shot, see what you get.